nortakales/flipper-zero-tonies #40: Toy Story 2

SamNitro, 2024-07-10:
Hey German/Disney/Toy Story 2.nfc ist fehlerhaft.

Gruß Patrick

knonymous83, 2024-07-10:
Hi, gerade gecheckt und die hier hinterlegte Datei ist tatsächlich fehlerhaft. Mein Original geht aber. Ich lade später erneut hoch

nortakales, 2024-07-11:
Sorry, don't speak German!

We'll need a new upload. It looks like a chunk of the file is actually missing.

I should implement some kind of validation in this repo. This kind of thing could be validated automatically (at least that all the correct fields are present).

SamNitro, 2024-07-11:
Okay, thanks for your great work.

knonymous83, 2024-07-11:
Hi, yeah I noticed that something is wrong with the uploaded info. I will correct that asap, but I noticed that I accidentally deleted my original file. I will rescan it asap.

knonymous83, 2024-07-12:
I have just updated the file and created a pull request.

deathduck666, 2024-07-12:
Corrected file works without any problems. Thanks for correction!
@nortakales Issue can be closed

closed 2024-07-12 by nortakales