nortakales/flipper-zero-tonies #93: Liste aller Tonies

cyberspy84, 2025-02-28:
Hallo
Kein Fehler oder sowas in der Art aber so langsam werden es doch schon einige Tonies hier. Gibt es eine Liste oder sowas wo alle drauf sind und man sieht welche hier schon gelistet sind und welche noch Fehlen? Dies würde den Doppelpost erleichtern.

Habe mal eine Liste im Internet gefunden die aber noch nicht Vollständig ist. Hab da mal ein paar markiert die ich hier in der List gefunden habe aber wie geagt nicht vollzählig.

Mit einer List wäre es auch einfacher zu sehen welche noch Fehlen.

Nur so eine Idee, wenn interesse besteht kann ich ja mal meine Liste hier einfügen und kann dann Erweitert werden, wenn interesse besteht natürlich nur.

Danke und echt Klasse was hier in kurzer Zeit zusammen gekommen ist :-)

felixlehmann, 2025-02-28:
Ganz allgemein lädt TeddyBench bei jedem Start eine aktuelle komplette Liste als JSON. 

Entsprechend des Quellcodes hier: https://github.com/toniebox-reverse-engineering/teddy/blob/03e1d6efffbf0353ea9899b901942f7aa28c49e3/TeddyBench/TeddyMain.cs#L151C85-L151C119

Ist das diese Datei: https://api.revvox.de/tonies.json

b2un0, 2025-02-28:
für ein mapping müsste man erstmal mit Hilfe der NFC Daten das Model raus finden. Das geht aber nicht soweit ich die Doku verstanden habe.
https://tonies-wiki.revvox.de/docs/wiki/general/protocol-analysis/#v2-content-get-v2contentuid-rev
hier wird ja die `UID` und der Auth Kram aus `Data Content` genommen.
Erst im Response steht drin, welche ModelID bzw `AudioID` es ist.

Einfacher wäre es dann wohl wenn man jede `.nfc` Datei mit der  `ModelID` oder `AudioID` oder prefixen.

z.b. 

alt: `German/Alea Aquarius/Die Magie der Nixen.nfc`
neu: `German/Alea Aquarius/1615480986_Die Magie der Nixen.nfc`

b2un0, 2025-02-28:
das Script hier generiert eine `KNOWN_TONIES.md` Tabelle, wenn es eine `.nfc` Datei mit dem vorgeschlagenem Muster gibt erscheint das ✅.

Problem ist, die ist `5570` Zeilen kan und somit unbrauchbar.

```bash
#!/usr/bin/env bash

set -e

TONIES_JSON_REMOTE=" https://raw.githubusercontent.com/toniebox-reverse-engineering/tonies-json/release/tonies.json"
TONIES_JSON_LOCAL="tonies.json.2"

if [ ! -f "${TONIES_JSON_LOCAL}" ]; then
  wget -O "${TONIES_JSON_LOCAL}" "${TONIES_JSON_REMOTE}"
fi

nfc_files=()
while IFS= read -r line; do
  nfc_files+=("$line")
done < <(find . -type f -regex '.*/[0-9]\{10\}_.*\.nfc')

{
echo "# Known Tonies"
echo ""
echo "> auto generated, do not edit manually"
echo ""
echo "| status | model | title | audio_id |"
echo "| ------ | ----- | ----- | -------- |"
} > KNOWN_TONIES.md

jq -r '.[] | [.audio_id[0] // "", .model, .title] | @tsv' "${TONIES_JSON_LOCAL}" | while IFS=$'\t' read -r audio model title; do
 status="❌"
  if [ -n "$audio" ]; then
      for file in "${nfc_files[@]}"; do
        fname=$(basename "$file")
        if [ "${fname:0:10}" = "$audio" ]; then
          status="✅"
          break
        fi
      done
    fi
  echo "| $status | $model | $title | $audio |" >> KNOWN_TONIES.md
done
```

das ist meiner Meinung nach zu viel.

cyberspy84, 2025-03-01:
War ja nur eine Idee das man eine kleine übersicht mal hat welche hier schon sind und welche noch fehlen.

Fabunator, 2025-03-01:
Ich hatte im Zusammenhang mit der json Liste aller Tonies eine andere Idee.
Ich habe ein Tool geschrieben was bei der Namensgebung und Einsortierung in den richtigen Ordner hilft, indem man aus einer Oberfläche den Tonie anhand von Sprache, Serie und Episode auswählen kann.
Die Datei wird dann automatisch benannt und verschoben.
Ich würde das in dieses repo hochladen, wenn gewünscht oder als eigenes Projekt.

nortakales, 2025-03-02:
I'd like to stay out of the business of keeping a database of ALL known Tonies. As mentioned, https://github.com/toniebox-reverse-engineering/tonies-json exists and I don't think we need to duplicate it here.

Love the idea of the markdown file listing all the Tonies we have in this repo though, let's definitely add that. Maybe we can find a creative way to get the photos and titles sourced from tonies.json, I'll give some more thought to that.

nortakales, 2025-03-02:
I just noticed something about tonies.json though, for some sets it does not have separate entries. For example, there are actually 4 Tonies in this set that would all be separate files for the Flipper:

```
{
        "no": "0",
        "model": "11000710",
        "audio_id": [
            "1739261703",
            "1739261429",
            "1739261328",
            "1739261189",
            "1730484293",
            "1730483684",
            "1730483371",
            "1730482970",
            "1710548949",
            "1710548884",
            "1710548850",
            "1710548815",
            "1695064912",
            "1695064819",
            "1695064754",
            "1695064693"
        ],
        "hash": [
            "EAEA6C05E5BAC352B3A6559B31AF3B0B9FD48F57",
            "D559116CA5A3EC2173E46FDB0EB8A4F0D338003F",
            "1451F41BF52CB9CA6241038FDC1B78158B292BA2",
            "7264A96447C542DD7EC04F38A18EE5050990F537",
            "8E97CF15282B6568D9D4F24B809E269280937940",
            "5ADEC3E5F3B90DF6A90678ED49538AF527228C83",
            "19767307BC30618A64DE08435D37C5660261FBFC",
            "D0298F81A72A9D07BE0BD90B713C68F8E302FDAD",
            "B11613BB33A6EA4443FF84741106A59DD742E206",
            "36AFC9EE152CD5BC842021C9D588B6748A1B0BF9",
            "D2FF164F4B847C5EDBEFE05B4E6DEF2EDEF63AA6",
            "46CD90740CA20411E90377FC02228E78EBDB368E",
            "E8730F2B57A2C9E737D435A2B5207F800975CDC5",
            "763580291D12F222CE0C2EE9389BCBAB28E5970D",
            "BBDC222BD06841B9CF0FF36DCF7262303B82B76F",
            "65CE9BAEAFCB371790B59DB14F5060F4CB730824"
        ],
        "title": "Clever Tonies Set - Inventions and Innovation",
        "series": "Clever Tonies Set",
        "episodes": "Inventions and Innovation",
        "tracks": [],
        "release": "1698169273",
        "language": "en-us",
        "category": "Tonie",
        "pic": "https://cdn.shopify.com/s/files/1/0403/5431/6439/files/inventions-main_1.jpg?v=1713268106"
    },
```

b2un0, 2025-03-03:
>Love the idea of the markdown file listing all the Tonies we have in this repo though, let's definitely add that. Maybe we can find a creative way to get the photos and titles sourced from tonies.json, I'll give some more thought to that.

i already tried this, test my bash scripts above. 

you also could let translate and explain ChatGTP what we are discuss here in german :D

nortakales, 2025-03-06:
> there is no good way to get the "real" AudioID or "modelID" from the .nfc files :(

I was really hoping this was possible, but I don't think either of those IDs are contained in the `.nfc` data at all. I was thinking maybe the first or last sections of hex may have been one of those IDs, but after spot checking some... nope.

So, we are left with possible just matching based on the filename, which even if people name them well probably needs to be a fuzzy match anyways because of variations in punctuation or other symbols, etc.

knonymous83, 2025-06-16:
> > Was hoping to get a full list of missing Tonies, too. That would make it easier to target farm them.
> 
> there are currently about ~ `5570` Tonies, are you really want to farm them all? :D

Actually yes or at least know a little faster, if the ones I see/find are not yet available. I got plenty of shops that have a lot of tonies.

FabulosoDev, 2025-09-03:
Maybe that collection of physical released tonies might help: https://fabulosodev.github.io/tonies-collection/

bluenazgul, 2025-09-03:
Simple reqeust to "ChatGPT 5-o thinking"

```
"If I have this JSON [https://raw.githubusercontent.com/toniebox-reverse-engineering/tonies-json/release/tonies.json](https://raw.githubusercontent.com/toniebox-reverse-engineering/tonies-json/release/tonies.json?utm_source=chatgpt.com)
, is it possible to build a script that uses it to rename all the NFC files from here [https://github.com/nortakales/flipper-zero-tonies](https://github.com/nortakales/flipper-zero-tonies?utm_source=chatgpt.com)
 so that they follow the format Model - Name.nfc instead of the current name.nfc?"
```

```
# Repo klonen (falls noch nicht geschehen)
git clone https://github.com/nortakales/flipper-zero-tonies.git
cd flipper-zero-tonies

```

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import argparse, json, re, sys, unicodedata, csv
from pathlib import Path
from urllib.request import urlopen, Request

# -----------------------------
# Utilities
# -----------------------------

def normalize(s: str) -> str:
    """Aggressive, aber robuste Normalisierung (ASCII, lower, no punctuation)."""
    s = unicodedata.normalize("NFKD", s).encode("ascii", "ignore").decode("ascii")
    s = s.lower()
    s = re.sub(r"[^\w\s-]", " ", s)   # Satzzeichen raus
    s = re.sub(r"[_-]+", " ", s)
    s = re.sub(r"\s+", " ", s).strip()
    return s

# (2a) Umlaut-Varianten: ae/oe/ue/ss <-> ä/ö/ü/ß
UMLAUT_MAPS = [
    ({"ae": "ä", "oe": "ö", "ue": "ü", "ss": "ß"}, "to_umlaut"),
    ({"ä": "ae", "ö": "oe", "ü": "ue", "ß": "ss"}, "from_umlaut"),
]

def umlaut_variants(s: str) -> set[str]:
    """Erzeuge beidseitige Umlaut-Varianten für einen String."""
    variants = {s}
    for mapping, _ in UMLAUT_MAPS:
        t = s
        for k, v in mapping.items():
            # ersetze case-insensitive
            t = re.sub(k, v, t, flags=re.IGNORECASE)
        variants.add(t)
    return variants

def tokens(s: str) -> set:
    return set(normalize(s).split())

def jaccard(a: set, b: set) -> float:
    if not a or not b:
        return 0.0
    inter = len(a & b)
    uni = len(a | b)
    return inter / uni if uni else 0.0

# Dateien mit bereits vorhandenem Modellpräfix überspringen
MODEL_PREFIX_RE = re.compile(r'^(\d{5,}|(\d{2}-\d{3,5}))\s+-\s+')

def has_model_prefix(stem: str) -> bool:
    """Prüft, ob der Dateiname (ohne .nfc) bereits mit einer Modellnummer beginnt."""
    return bool(MODEL_PREFIX_RE.match(stem))

[... 221 lines cut ...]
                    grandparent = up.parent.parent.name if up.parent and up.parent.parent else ""
                    w.writerow([u, up.name, parent, grandparent])
            print(f"\nCSV mit Unmatched geschrieben: {report_path}")
        except Exception as e:
            print(f"[WARN] Konnte CSV nicht schreiben: {e}", file=sys.stderr)

    print(f"\nFertig. Umbenannt: {renamed}, übersprungen: {skipped}, ohne Treffer: {len(unmatched)}")

if __name__ == "__main__":
    main()
```

# save script somewhere
cd ..
python3 rename_tonies.py ./flipper-zero-tonies \
  --json "https://raw.githubusercontent.com/toniebox-reverse-engineering/tonies-json/release/tonies.json" \
  --dry-run

# if everythin works well, run without --dry-run
python3 rename_tonies.py ./flipper-zero-tonies \
  --json "https://raw.githubusercontent.com/toniebox-reverse-engineering/tonies-json/release/tonies.json"

for me it worked very well with only 62 unmatching files

the script also creates an _rename_unmatched.csv_ that includes all unmatching files

FabulosoDev, 2025-09-03:
no need to do that fuzzy match.
here is the list you are looking for:
https://gist.githubusercontent.com/FabulosoDev/8b772d5a4cb39e985fa9c3c06d001f65/raw/0044d3c4e57519aad9375ad32857c68245070705/flipper-zero-tonies.json

FabulosoDev, 2025-09-04:
based on the json I mentioned you don't need to do the fuzzy matching in your script.
just get model and title based on file from that json and rename it according to your normalize(s: str) function

nortakales, 2025-09-09:
Why do you want to rename the files to have model numbers? I'd prefer for them to not have the model numbers, because the file names are what you end up seeing in menus when put these on a Flipper and emulate them. Some arbitrary number would be useless at that point.

Iind0s, 2025-09-13:
Kann man die slix-L überschreiben?