nortakales/flipper-zero-tonies #124: Sind die Tonies unique?

skydiablo, 2025-08-26:
Moin!
Sind die Tonies eigentlich unique? also ein und die gleiche Geschichte auf verschiedenen Tonie Figuren haben irgendwelche unterschiede? Ist dort eine ID unterschiedlich oder im Payload noch ein vortlaufender "key" versteckt?

Also kann man mit dem scan sagen, genau dieser Tonie steht bei mir im Regal und ist quasi meinem Account zuzuordnen?

Grüße, Volker.

deathduck666, 2025-08-26:
Hallo Volker,

die Tonies unterscheiden sich eigentlich nur in 2 Punkten:

1. UID: quasi die ID des Tonies. Diese sollte auch für die Verknüpfung in deinem Tonies Account verantwortlich sein. 
--> Toniebox prüft bei erstmaligem Draufstellen, ob die UID in der Tonie Datenbank hinterlegt ist
2. Data Conent: Dahinter verbirgt sich wie der Name schon verrät, der Content. Also welche Figur steht eigentlich auf der Box.

Einen fortlaufenden "key" (meinst du so etwas wie einen Rolling Code bei Schlüsseln?) gibt es nicht.
Die Tonies werden ja u.a. auch in Bibliotheken an viele „Haushalte“ verliehen.

Grüße

b2un0, 2025-08-27:
Genau. Jeder tonie ist einzigartig weltweit. 
Bedeutet aber auch, dass wir hier unzählige Kopien haben und diese IDs bei Boxine in der DB womöglich mehrfach pro Tag einem anderen Account zugeordnet werden. 

Theoretisch also könnte Boxine Tonies sperren welche wir hier listen. Theoretisch. Machen sie ja aber seit Jahren nicht.

nortakales, 2025-08-28:
Looks like your question was answered. I'll close the issue for now.

Google translate:
> Ihre Frage scheint beantwortet worden zu sein. Ich schließe das Problem vorerst.

closed 2025-08-28 by nortakales